IntergatedCircuits/USBDevice4Cube #1: Various issues in USB_vAllocateEPs

igorizyumin, 2018-11-17:
The USB_vAllocateEPs function in xpd_usb_otg.c does not set the FIFO sizes correctly, at least for the STM32F4xx series MCUs.  The FIFO depth registers are in 32b words, but the code treats it as if it was in bytes.  This results in overlapping FIFOs getting allocated, with catastrophic results.  

Also, the global RX FIFO needs to be larger than the maximum packet size by one word.  Otherwise, the USB core NAKs all incoming packets.

IntergatedCircuits, 2018-11-17:
> The FIFO depth registers are in 32b words, but the code treats it as if it was in bytes.

Where do you get this idea from? In `USB_vAllocateEPs()`:
```C
    /* Global RX FIFO */
    pxUSB->Inst->GRXFSIZ = ulFifoSize;

    /* Fifo sizes should be word aligned */
    ulMPS = (pxUSB->EP.IN[0].MaxPacketSize + 3) & (~3);

    /* EP0 TX FIFO */
    pxUSB->Inst->DIEPTXF0_HNPTXFSIZ.w =
            (ulMPS      << USB_OTG_DIEPTXF_INEPTXFD_Pos) |
            (ulFifoSize << USB_OTG_DIEPTXF_INEPTXSA_Pos);

    /* Increase offset with the FIFO size */
    ulFifoSize += pxUSB->EP.IN[0].MaxPacketSize;

    for (ucEpNum = 1; ucEpNum < ucEpCount; ucEpNum++)
    {
        /* Fifo sizes should be word aligned */
        ulMPS = (pxUSB->EP.IN[ucEpNum].MaxPacketSize + 3) & (~3);

        /* EPx TX FIFOs */
        pxUSB->Inst->DIEPTXF[ucEpNum - 1].w =
                (ulMPS      << USB_OTG_DIEPTXF_INEPTXFD_Pos) |
                (ulFifoSize << USB_OTG_DIEPTXF_INEPTXSA_Pos);

        /* Increase offset with the current FIFO size */
        ulFifoSize += pxUSB->EP.IN[ucEpNum].MaxPacketSize;
    }
```
And in `__USB_OTG_TypeDef`:
```C
    __IO uint32_t GRXFSIZ;
```
```C
    union {
        struct {
            __IO uint32_t INEPTXSA : 16; 
            __IO uint32_t INEPTXFD : 16;
        } b;
        __IO uint32_t w;
    } DIEPTXF0_HNPTXFSIZ;
```
```C
    union {
        struct {
            __IO uint32_t INEPTXSA : 16;
            __IO uint32_t INEPTXFD : 16;
        } b;
        __IO uint32_t w;
    } DIEPTXF[5];
```

IntergatedCircuits, 2018-11-17:
> Also, the global RX FIFO needs to be larger than the maximum packet size by one word.

According to the reference manual, you are indeed correct:

> One location is to be allocated for Global OUT NAK. Status information
is written to the FIFO along with each received packet. Therefore, a minimum space of
(Largest Packet Size / 4) + 1 must be allocated to receive packets. If multiple isochronous
endpoints are enabled, then at least two (Largest Packet Size / 4) + 1 spaces must be
allocated to receive back-to-back packets. Typically, two (Largest Packet Size / 4) + 1
spaces are recommended so that when the previous packet is being transferred to the CPU,
the USB can receive the subsequent packet.

However, ST's OTG documentation is famously useless. I have also failed to find a single example in the Cube example projects that adheres to this rule. But the most convincing argument to emit this extra word was that I tested the code myself (with STM32F407 and STM32L476 OTG_FS), and it works perfectly fine for me.

I did foresee that this fixed way of FIFO allocation isn't suitable for all use-cases, that's why it's defined with the `__weak` attribute. So you can write your own project-specific version of the function, in which case your function will be linked instead of the default.

igorizyumin, 2018-11-17:
Hi, sorry, I should have been more clear.  I am using the HS core on a 429 chip (same as the 407 for our purposes).  The issue I have is with this line (and the corresponding line for EP0):
```
        pxUSB->Inst->DIEPTXF[ucEpNum - 1].w =
                (ulMPS      << USB_OTG_DIEPTXF_INEPTXFD_Pos) |
                (ulFifoSize << USB_OTG_DIEPTXF_INEPTXSA_Pos);
```
ulMPS is in bytes, but the core expects that value to be in 32-bit words.  Therefore, the FIFO depth is getting configured 4x larger than it should be, and so the various FIFOs are overlapping.  Surprisingly, this can actually function for some time; I could only provoke the bug by filling up the Rx FIFO with a bunch of small transactions while sending large blocks from a Tx endpoint.  This would result in the hardware getting completely hosed up and sending completely invalid packets (wrong CRC, etc).

From the reference manual RM0090 (page 1429):

> Bits 31:16 INEPTXFD: IN endpoint TxFIFO depth
> This value is in terms of 32-bit words.
> Minimum value is 16
> Maximum value is 512

As for the other issue, I can show you a USB analyzer trace where the USB core was NAKing 512-byte packets when the Rx FIFO was configured for 512 bytes, so the documentation is correct.  You are not seeing this issue because of the first problem (the FIFO is getting configured as 4x larger than the MPS).

IntergatedCircuits, 2018-11-17:
I see the problem now. I tried to implement the function based solely on the instructions of the reference manual, but that only made things worse. In the end I managed to create a proper setup that is working properly on my side:

```C
__weak void USB_vAllocateEPs(USB_HandleType * pxUSB)
{
    XPD_ReturnType eResult = XPD_OK;
    uint8_t ucEpNum;
    uint8_t ucEpCount = USB_ENDPOINT_COUNT(pxUSB);
    uint32_t ulMinFifoSizeVal = 16;
    uint32_t ulFifoSize = ulMinFifoSizeVal * sizeof(uint32_t);
    uint32_t ulFifoOffset;
    uint32_t ulFifoLimit = USB_TOTAL_FIFO_SIZE(pxUSB);

#ifdef USB_OTG_HS
    if (USB_eDevSpeed(pxUSB) == USB_SPEED_FULL)
    {
        /* Full-Speed endpoint sizes should be limited first */
        for (ucEpNum = 0; ucEpNum < ucEpCount; ucEpNum++)
        {
            USB_prvLimitEpMPS_FS(&pxUSB->EP.IN [ucEpNum]);
            USB_prvLimitEpMPS_FS(&pxUSB->EP.OUT[ucEpNum]);
        }
    }
#endif

    /* Configure the global Receive FIFO as the largest requested OUT EP size */
    for (ucEpNum = 0; ucEpNum < ucEpCount; ucEpNum++)
    {
        if (pxUSB->EP.OUT[ucEpNum].MaxPacketSize > ulFifoSize)
        {
            ulFifoSize = pxUSB->EP.OUT[ucEpNum].MaxPacketSize;
        }
    }

    /* Fifo sizes are in words */
    ulFifoSize = (ulFifoSize + 3) >> 2;

    /* Global RX FIFO */
    ulFifoOffset = (4 + 6) + (ulFifoSize + 1) + (ucEpCount * 2) + 1;
    pxUSB->Inst->GRXFSIZ = ulFifoOffset;

    /* EP0 TX FIFO */
    ulFifoSize = (pxUSB->EP.IN[0].MaxPacketSize + 3) >> 2;

    pxUSB->Inst->DIEPTXF0_HNPTXFSIZ.w =
            (ulFifoSize   << USB_OTG_DIEPTXF_INEPTXFD_Pos) |
            (ulFifoOffset << USB_OTG_DIEPTXF_INEPTXSA_Pos);

    for (ucEpNum = 1; ucEpNum < ucEpCount; ucEpNum++)
    {
        /* Increase offset with the FIFO size */
        ulFifoOffset += ulFifoSize;

        /* Fifo sizes should be word aligned */
        ulFifoSize = (pxUSB->EP.IN[ucEpNum].MaxPacketSize + 3) >> 2;
        if (ulFifoSize < ulMinFifoSizeVal)
        {   ulFifoSize = ulMinFifoSizeVal; }

        /* EPx TX FIFOs */
        pxUSB->Inst->DIEPTXF[ucEpNum - 1].w =
                (ulFifoSize   << USB_OTG_DIEPTXF_INEPTXFD_Pos) |
                (ulFifoOffset << USB_OTG_DIEPTXF_INEPTXSA_Pos);
    }

    /* Total FIFO use shouldn't exceed available size */
    if (ulFifoLimit < (ulFifoOffset + ulFifoSize))
    {
        eResult = XPD_ERROR;
    }

    (void)eResult;
}
```

Could you check if this fixes your issues on OTG_HS as well?

IntergatedCircuits, 2018-11-17:
I had a look at the OTG Rx FIFO configuration documentation of different STM32 series, and it's a total mess. In all cases, the text is pretty much the same:
> Receive FIFO RAM allocation: the application should allocate RAM for SETUP Packets:
• 10 locations must be reserved in the receive FIFO to receive SETUP packets on control endpoint. The core does not use these locations, which are reserved for SETUP packets, to write any other data.
> • One location is to be allocated for Global OUT NAK.
> • Status information is written to the FIFO along with each received packet. Therefore, a minimum space of (Largest Packet Size / 4) + 1 must be allocated to receive packets. If multiple isochronous endpoints are enabled, then at least two (Largest Packet Size / 4) + 1 spaces must be allocated to receive back-to-back packets. Typically, two (Largest Packet Size / 4) + 1 spaces are recommended so that when the previous packet is being transferred to the CPU, the USB can receive the subsequent packet.
> • Along with the last packet for each endpoint, transfer complete status information is also pushed to the FIFO. One location for each OUT endpoint is recommended.

Beyond that, the 'F4 ref-man has no further description. The 'F7 ref-man adds this formula as a representation of the above text:
>(4 * number of control endpoints + 6) + ((largest USB packet used / 4) + 1 for status information) + (2 * number of OUT endpoints) + 1 for Global NAK

While the 'L4 ref-man adds this formula for the same:
>(5 * number of control endpoints + 8) + ((largest USB packet used / 4) + 1 for status information) + (2 * number of OUT endpoints) + 1 for Global NAK

Both formulas contradict the text, and each other as well. I'm leaning towards using the 'F7 formula, where in the end the only unexplained part is the 2 multiplier.

IntergatedCircuits, 2018-11-20:
I'm glad to hear that the problem is solved. You are welcome to report any other issues, especially with the DMA of the OTG_HS and data alignments, as I have no such hardware to test it with.

closed 2018-11-20 by IntergatedCircuits