IntergatedCircuits/USBDevice4Cube #4: stm32f407  NOVBUSSENS not defined

Warloxx, 2019-05-26:
I came across a compilation error when using this USB Stack on the stm32f407vg (discovery board).
Since the stm32f407 does not declare the symbol USB_OTG_GCCFG_VBDEN on line 718 in file XPD_USB/xpd_usb_otg.c you try to set the field NOVBUSSENS of register GCCFG. But this field does not exist. In file XPD_USB/xpd_config.h line 1004 the field that is probably meant to be set has a different name "VBDEN".
Renaming that field in xpd_config.h to NOVBUSSENS compiles and an initial test also works during runtime.

As a reference the HAL uses this (stm32f4xx_ll_usb.c:208)
```c
#if defined(STM32F446xx) || defined(STM32F469xx) || defined(STM32F479xx) || defined(STM32F412Zx) || defined(STM32F412Vx) || \
    defined(STM32F412Rx) || defined(STM32F412Cx) || defined(STM32F413xx) || defined(STM32F423xx)
```
to distinguish between the two field names. They might actually have slightly different functionalities, but I'm not sure.
The datasheet for the 401/405/415/407/411/417/427/437/429/439 definitely use the name NOVBUSSENS, and also other names for the other fields (except PWRDWN) and bit 17&3:0 are reserved.
While the datasheet for the 412/413/423 describe the register fields as described in xpd_config.h.
The 446/469/479 apparently only has bits 21 (VBDEN) and 16 (PWRDWN).

I made the effort of collecting every different layout of the GCCFG register through all the families (see below). I'm not sure if NOVBUSSENS  and VBDEN are really the same but just a different name. (All the other different fields seem to only be used during the charger detection which gets disabled if the fields don't exist.) So if it is the same name then maybe just use VBDEN at XPD_USB/xpd_usb_otg.c:718. 

```c
    union {
        struct {

#if defined(STM32F105xC) || defined(STM32F107xC)
            /* This register layout is shared by these models:
             * F105/107
             */
                 uint32_t __RESERVED0 : 16;
            __IO uint32_t PWRDWN : 1;                       /*!< Power down */
                 uint32_t __RESERVED1 : 1;
            __IO uint32_t VBUSASEN : 1;                     /*!< Enabled Vbus sensing A Device */
            __IO uint32_t VBUSBSEN : 1;                     /*!< Enabled Vbus sensing B Device */
            __IO uint32_t SOFOUTEN : 1;                     /*!< SOF output enable */
                 uint32_t __RESERVED2 : 11;
#elif defined(STM32F205xx) || defined(STM32F207xx) || defined(STM32F215xx) \
   || defined(STM32F217xx)
            /* This register layout is shared by these models:
             * F205/207/215/217
             * */
                 uint32_t __RESERVED0 : 16;
            __IO uint32_t PWRDWN : 1;                       /*!< Power down */
            __IO uint32_t I2CPADEN : 1;                     /*!< Battery charging detector (BCD) enable */
            __IO uint32_t VBUSASEN : 1;                     /*!< Enabled Vbus sensing A Device */
            __IO uint32_t VBUSBSEN : 1;                     /*!< Enabled Vbus sensing B Device */
            __IO uint32_t SOFOUTEN : 1;                     /*!< SOF output enable */
                 uint32_t __RESERVED1 : 11;
#elif defined(STM32F401xC) || defined(STM32F401xE) || defined(STM32F405xx) \
   || defined(STM32F415xx) || defined(STM32F407xx) || defined(STM32F411xE) \
   || defined(STM32F417xx) || defined(STM32F427xx) || defined(STM32F437xx) \
   || defined(STM32F429xx) ||defined(STM32F439xx)
           /* This register layout is shared by these models:
            * F401/405/415/407/411/417/427/437/429/439
            */
                uint32_t __RESERVED0 : 16;
           __IO uint32_t PWRDWN : 1;                       /*!< Power down */
                uint32_t __RESERVED1 : 1;
           __IO uint32_t VBUSASEN : 1;                     /*!< Enabled Vbus sensing A Device */
           __IO uint32_t VBUSBSEN : 1;                     /*!< Enabled Vbus sensing B Device */
           __IO uint32_t SOFOUTEN : 1;                     /*!< SOF output enable */
           __IO uint32_t NOVBUSSENS : 1;                   /*!<  Vbus sensing disable option*/
                uint32_t __RESERVED2 : 10;
#elif defined(STM32F412Cx) || defined(STM32F412Zx) || defined(STM32F412Rx) \
   || defined(STM32F412Vx) || defined(STM32F413xx) || defined(STM32F423xE) \
   \
   || defined(STM32H743xx) || defined(STM32H750xx) || defined(STM32H753xx) \
   \
   || defined(STM32L4A6xx) || defined(STM32L4R5xx) || defined(STM32L4R7xx) \
   || defined(STM32L4R9xx) || defined(STM32L4S5xx) || defined(STM32L4S7xx) \
   || defined(STM32L4S9xx) || defined(STM32L475xx) || defined(STM32L476xx) \
   || defined(STM32L485xx) || defined(STM32L486xx) || defined(STM32L496xx)
           /* This register layout is shared by these models:
            * F412/413/423
            * 4a6/4r5/4r7/4r9/4s5/4s7/4s9/475/476/485/486/496
            * H743/750/753
            */
           __IO uint32_t DCDET : 1;                        /*!< Data contact detection (DCD) status */
           __IO uint32_t PDET : 1;                         /*!< Primary detection (PD) status */
           __IO uint32_t SDET : 1;                         /*!< Secondary detection (SD) status */
           __IO uint32_t PS2DET : 1;                       /*!< DM pull-up detection status */
                uint32_t __RESERVED0 : 12;
           __IO uint32_t PWRDWN : 1;                       /*!< Power down */
           __IO uint32_t BCDEN : 1;                        /*!< Battery charging detector (BCD) enable */
           __IO uint32_t DCDEN : 1;                        /*!< Data contact detection (DCD) mode enable*/
           __IO uint32_t PDEN : 1;                         /*!< Primary detection (PD) mode enable*/
           __IO uint32_t SDEN : 1;                         /*!< Secondary detection (SD) mode enable */
           __IO uint32_t VBDEN: 1;                         /*!< Vbus detection enable */
                uint32_t __RESERVED2 : 10;
#elif defined(STM32F723xx) || defined(STM32F730xx) || defined(STM32F733xx)
           /* This register layout is shared by these models:
            * F723/730/733
            */
           __IO uint32_t DCDET : 1;                        /*!< Data contact detection (DCD) status */
           __IO uint32_t PDET : 1;                         /*!< Primary detection (PD) status */
           __IO uint32_t SDET : 1;                         /*!< Secondary detection (SD) status */
           __IO uint32_t PS2DET : 1;                       /*!< DM pull-up detection status */
                uint32_t __RESERVED0 : 12;
           __IO uint32_t PWRDWN : 1;                       /*!< Power down */
           __IO uint32_t BCDEN : 1;                        /*!< Battery charging detector (BCD) enable */
           __IO uint32_t DCDEN : 1;                        /*!< Data contact detection (DCD) mode enable*/
           __IO uint32_t PDEN : 1;                         /*!< Primary detection (PD) mode enable*/
           __IO uint32_t SDEN : 1;                         /*!< Secondary detection (SD) mode enable */
           __IO uint32_t VBDEN: 1;                         /*!< Vbus detection enable */
           __IO uint32_t OTGIDEN: 1;
           __IO uint32_t PHYHSEN: 1;
                uint32_t __RESERVED2 : 8;
#elif defined(STM32F722xx) || defined(STM32F732xx) || defined(STM32F745xx) \
   || defined(STM32F746xx) || defined(STM32F750xx) || defined(STM32F756xx) \
   || defined(STM32F765xx) || defined(STM32F767xx) || defined(STM32F769xx) \
   || defined(STM32F777xx) || defined(STM32F779xx)
           /* This register layout is shared by these models:
            * F722/732/745/746/750/756/765/767/769/777/779
            */
                uint32_t __RESERVED0 : 16;
           __IO uint32_t PWRDWN : 1;                       /*!< Power down */
                uint32_t __RESERVED1 : 4;
           __IO uint32_t VBDEN : 1;                        /*!< Vbus detection enable */
                uint32_t __RESERVED2 : 10;
#else
#error "Unsupported device"
#endif
        } b;
        __IO uint32_t w;
    } GCCFG;                                                /* General Purpose IO Register                     038h*/
```

IntergatedCircuits, 2019-05-28:
Thanks for the very thorough investigation into this issue, there is always something to learn about this module. The outrageous thing is that the NOVBUSSENS and the VBDEN bits correspond to the same location, but their logic is the opposite of one another, so they definitely shouldn't be confused.

I chose to add all known fields to either variant of this register, as the code anyway has precompile checks to see which bits are defined. Writing to a reserved bit when setting NOVBUSSENS on a machine that doesn't have either of the two bits should have no side effects.

Warloxx, 2019-05-30:
That is really nasty that they have opposite behavior.
I actually have to admit that the layout I posted earlier is flawed. Some of that information was initially parsed from the datasheets but then I switched to CMSIS files since they are easier to search. This turned out to be a bad call since they are not complete. As I later found out most of the F7xx CMSIS files only define the PWRDWN and VBDEN fields. This is somewhat correct. Those are the one available for the HS version while the FS version is like the version from the  F412/413/423 - L4a6/4r5/4r7/4r9/4s5/4s7/4s9/475/476/485/486/496 - H743/750/753 group. But that was all just missing.

closed 2019-05-30 by IntergatedCircuits

Ultrawipf, 2020-01-22:
Is there a fix now? I am trying to get it working on an STM32F411 which also has the VBDEN field but no NOVBUSSENS.
Changing it to VBDEN and reversing the logic i can't get it to connect at all. Having VBDEN on it does connect but i can't get the CDC example to work (Descriptor invalid...).

IntergatedCircuits, 2020-01-23:
What is your current issue without touching the code? If it's an invalid descriptor then it has nothing to do with this bit.

Ultrawipf, 2020-01-23:
It was the console_if example with pretty much nothing changed apart from the VBDEN issue and a basic ST-Cube setup. Not sure if i missed something but it was pretty much set up according to the short readme of USBDevice with the wrapper instead of the ST usb functions.

I currently have a setup with the ST HAL based on their middlewares but a more modular composite device with bidirectional HID and CDC would be great.

IntergatedCircuits, 2020-01-24:
You can open a new issue if you can record the USB packets, then I can take a look at what could have gone wrong.