IntergatedCircuits/USBDevice4Cube #5: F7 not enough space in default FIFO allocation

Warloxx, 2019-05-30:
I setup a project with cube and freertos on an F767ZI (nucleo board) with one ACM CDC using the supplied console_if and it was not enumerating while recreating the same project worked fine on a F407 (after fixing #4). Both times using the FS OTG,

After a lot of head-scratching I just added a few more words to the FIFO in USB_HvAllocateEPs.
It turned out that just **one** more word was the difference between working and not working.
After reading #1 I found out that the quoted app-note in the comment RM0431 uses a slightly different formula then the one actually used and also clashes with the text description:
> 10 locations must be reserved in the receive FIFO to receive SETUP packets on control endpoint. [...]
(5 * number of control endpoints + 8) + [...]

how is this supposed to add up to 10? (also can you have more then one control EP?)

However the formula that is used matches the one that is documented in the doc for the F76x/F77x (RM0410).
> (4 * number of control endpoints + 6) + [...]

 I now share your frustration of unclear and nebulous documentation. 
By the way the CubeHAL just uses 128 words for both F4 and F7 no matter the configuration (512 when using HS).

I don't really know what to do about it. I will just add some extra words to the FIFO for now on my end. Since we apperently don't know what exact number of words is required it is a bit difficult to fix the default.

Warloxx, 2019-05-30:
I almost forgot. To make it work on the F7 I had to define the following symbols myself since apperently the F7 CMSIS files don't provide those :disappointed: 

```c
/*values from datasheet for 76xxx & 77xxx */
#define USB_OTG_FS_MAX_IN_ENDPOINTS       5 
#define USB_OTG_HS_MAX_IN_ENDPOINTS       8
#define USB_OTG_FS_TOTAL_FIFO_SIZE        1280
#define USB_OTG_HS_TOTAL_FIFO_SIZE        4096
```

IntergatedCircuits, 2019-06-06:
I don't have a better idea than just keep adjusting the value until everyone's happy, so I increased the default RxFIFO size by 1 word, if this doesn't cause further issues please close this ticket. 

For the missing symbols from the device header I can only recommend that you report it on the STM32 forum.

closed 2019-07-22 by IntergatedCircuits