IntergatedCircuits/USBDevice4Cube #7: IAR project generated by Cube for Nucleo-L4R5ZI enumeration issue

LACoopConsulting, 2019-10-21:
This project looks fantastic, especially since ST doesn't seem willing to put composite support in their libs! I'm running into an issue, unfortunately, with getting a starter project using this up and running.

I have taken the following steps:

1. Generated a  Nucleo-STM32L4R5ZI IAR project using CubeMX. I left all settings alone and didn't enable anything extra.
2. Added the latest XPD_USB from this repo's master branch, latest USBDevice from its repo's master branch to my IAR project, and each relevant include directory.
3. Added usb_device.c/h, stm32xxxx.h, hal_usb.c, main.c, and usb_device.c from this repo
4. Added usbd_config.h and console_if.c templates from USBDevice repo
5. Built, flashed, connected USB from my windows PC to my board.

After doing this, an entry appears in Device Manager under Universal Serial Bus controllers with this name `Unknown USB Device (Invalid Configuration Descriptor)`

Feels like there's something silly I may have missed. I checked into alignment pragmas but had no luck there. Any suggestions would be greatly appreciated, as I'd really like to use this in my projects. Thanks in advance!

IntergatedCircuits, 2019-10-22:
I'd recommend generating a USB project with Cube, and cross-checking the clock configuration, PWR module calls, etc as a first step.

LACoopConsulting, 2019-10-22:
Thanks for that info. I've confirmed the clock config is set to generate a 48MHz clock for the USB. Would it be helpful for me to make a private repo with my project? If so I'll get one put together and send you an invite.

IntergatedCircuits, 2019-10-24:
I just realized that the problem is with the Configuration Descriptor (usually the Device Descriptor fails the first time, as that's requested first). In this case I'd need to see what the USB device is sending to the host PC as the Config Descriptor (as a byte array) in order to tell more.

LACoopConsulting, 2019-10-25:
This zip has a csv file exported from TotalPhase's tool that I used with a Beagle 12 USB protocol analyzer. I believe it has all the bytes sent. Please let me know if this is sufficient. Thanks!

[l4r5zi_capture.zip](https://github.com/IntergatedCircuits/USBDevice4Cube/files/3769973/l4r5zi_capture.zip)

IntergatedCircuits, 2019-10-25:
I'm 95% confident that the issue is with the IAR not handling the __packed syntax as it should. You could try to compile and flash the same sources with GCC and cross-check the result.

LACoopConsulting, 2019-10-29:
Compiling with GCC did resolve the configuration descriptor and I was able to have a single console_if. Any advice for adding a second CDC? I've tried making another interface and mounting it, but I'm still only seeing one appear. This is with GCC compilation btw. If you'd rather I make a separate issue for this happy to close this one out and make a new one. Thanks.

IntergatedCircuits, 2019-10-29:
For the second CDC the only recommendation I have is to use endpoints that are valid and exist on the hardware. If it doesn't work with it, you can open a new issue.

closed 2019-10-29 by IntergatedCircuits