IntergatedCircuits/USBDevice4Cube #8: Is this suitable for a template-generated CubeIDE v1.1.0 project?

fadedbee, 2020-01-20:
I'd like to try https://github.com/IntergatedCircuits/USBDevice with a template-generated project in ST's CubeIDE.

The folder structure here is different (main.c is in Core/Src in my CubeIDE projects).

Is this wrapper project (and its instructions) suitable for CubeIDE v1.1.0?

fadedbee, 2020-01-20:
I've uploaded my brand-new CubeIDE project, so that you can see the folder structure.

https://github.com/fadedbee/f3-usbdevice

IntergatedCircuits, 2020-01-21:
Other than new folder structure, I don't see that much change in the Cube project, so it shouldn't be complicated to perform the integration.

fadedbee, 2020-01-22:
Thanks, I'll give it a go if I can't resolve https://stackoverflow.com/questions/59856541/cubemx-generated-usb-hid-device-sends-wrong-data-when-both-endpoint-and-pma-addr

closed 2020-01-22 by fadedbee

IntergatedCircuits, 2020-01-22:
Maybe you'll have better luck by posting the issue to ST Forum instead. The USBDevice library allocates PMA/FIFO automatically at Setup, based on the endpoint packet size configuration (see `USB_vAllocateEPs()`).