IntergatedCircuits/USBDevice4Cube #10: How to use with STM32CubeIDE

MickGyver, 2020-11-24:
This project looks promising, I'm not having any luck using it iwith STM32CubeIDE 1.4.0. Could you please give some hints what files are needed and where? I guess I need both this wrapper project and the USBDevice project but I don't know how to set it all up...

MickGyver, 2020-11-24:
I copied files from the two repositories and was able to compile the project but with warnings that I need to solve. For example this one in HAL_USBD_Setup():
```../Core/Src/hal_usb.c:34:36: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
     UsbDevice->Callbacks.DepDeinit = HAL_PCD_MspDeInit;```

MickGyver, 2020-11-24:
I guess what I'm not really understanding is how to setup the hal_usb.c file for STM32F103, since things seem quite different from your example.

benedekkupper, 2020-11-24:
It's best if you generate Cube code for the USB peripheral, then copy the related MspInit and MspDeInit calls to hal_usb.c, replace the OTG_FS_IRQHandler and OTG_FS_IRQn to the F103's USB interrupt vector.

For proper operation you'll need to connect 1.5K resistor between USB DP and a free GPIO output, and implement a UsbDevice->Callbacks.ConnectCtrl callback as well, to set this pin depending on the parameter.

MickGyver, 2020-11-26:
Thanks for your reply. I'm using Blue Pill boards that have a 1.5K resistor between USB DP and 3.3V. This is the code I have in hal_usb.c now. It gives me some warnings and the device is not recognised when plugged into a usb port. I'm still missing something I guess.

`void HAL_PCD_MspInit(PCD_HandleTypeDef* pcdHandle)
{
  if(pcdHandle->Instance == USB)
  {
    __HAL_RCC_USB_CLK_ENABLE();
    HAL_NVIC_SetPriority(USB_LP_CAN1_RX0_IRQn, 0, 0);
    HAL_NVIC_EnableIRQ(USB_LP_CAN1_RX0_IRQn);
  }
}

void HAL_PCD_MspDeInit(PCD_HandleTypeDef* pcdHandle)
{
  if(pcdHandle->Instance == USB)
  {
    __HAL_RCC_USB_CLK_DISABLE();
    HAL_NVIC_DisableIRQ(USB_LP_CAN1_RX0_IRQn);
  }
}

void HAL_USBD_Setup(void)
{
    USB_INST2HANDLE(UsbDevice, USB);
    UsbDevice->Callbacks.DepInit = HAL_PCD_MspInit;
    UsbDevice->Callbacks.DepDeinit = HAL_PCD_MspDeInit;
}

void USB_LP_CAN1_RX0_IRQHandler(void);

void USB_LP_CAN1_RX0_IRQHandler(void)
{
  USB_vIRQHandler(UsbDevice);
}`

benedekkupper, 2020-11-30:
I meant to copy the statements in these functions, the parameter type should be void*, and the if should be removed.
In general these should do 3 things:
1. Set the gpios to usb alternate mode
2. Set the clock tree of the usb so it gets clocked at 48 MHz
3. Configure the usb interrupt(s)

So you'll have to get these right at least for your usb to work.

MickGyver, 2020-11-30:
Thanks you I figured yesterday that only the statements needed copying, I first thought that the entire functions needed to be implemented. have it working now, below is how hal_usb.h looks now. (it worked before configuring the GPIO pins also, but wouldn't rely on it).

`
void HAL_USB_MspInit(void * handle)
{
	GPIO_InitTypeDef GPIO_InitStruct;
	GPIO_InitStruct.Pin = GPIO_PIN_11 | GPIO_PIN_12;
	GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
	GPIO_InitStruct.Pull = GPIO_NOPULL;
	GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
	HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);

	__HAL_RCC_USB_CLK_ENABLE();
	HAL_NVIC_SetPriority(USB_LP_CAN1_RX0_IRQn, 0, 0);
	HAL_NVIC_EnableIRQ(USB_LP_CAN1_RX0_IRQn);
}

void HAL_USB_MspDeInit(void * handle)
{
	HAL_GPIO_DeInit(GPIOA, GPIO_PIN_11 | GPIO_PIN_12);

	__HAL_RCC_USB_CLK_DISABLE();
	HAL_NVIC_DisableIRQ(USB_LP_CAN1_RX0_IRQn);
}

void HAL_USBD_Setup(void)
{
    USB_INST2HANDLE(UsbDevice, USB);
    UsbDevice->Callbacks.DepInit = HAL_USB_MspInit;
    UsbDevice->Callbacks.DepDeinit = HAL_USB_MspDeInit;
}
`

The IRQ handler is implemented in stm32f1xx_it.c:

`
void USB_LP_CAN1_RX0_IRQHandler(void)
{
  /* USER CODE BEGIN USB_LP_CAN1_RX0_IRQn 0 */

  /* USER CODE END USB_LP_CAN1_RX0_IRQn 0 */
  //HAL_PCD_IRQHandler(&hpcd_USB_FS);
  /* USER CODE BEGIN USB_LP_CAN1_RX0_IRQn 1 */
  USB_vIRQHandler(UsbDevice);
  /* USER CODE END USB_LP_CAN1_RX0_IRQn 1 */
}
`

closed 2020-11-30 by MickGyver

DCNick3, 2021-12-08:
It seems that STM32F103C8T6 has something funky going on with its GPIO config: CubeMX doesn't seem generate GPIO configuration for USB at all. Maybe it's because it's in the required mode for USB already at boot?

benedekkupper, 2021-12-08:
> Maybe it's because it's in the required mode for USB already at boot?
I'd rather guess that it doesn't generate USB code because you didn't select the USB_DEVICE in Middlewares. Which is fine if you are going to use this stack instead. From the above discussion you shouldn't have any problem adding the necessary glue code to make this work.