IntergatedCircuits/USBDevice4Cube #13: STM32L151 Integration

TannerHollis, 2023-01-14:
We are writing firmware for a custom HID device that uses the standard HAL libraries, but are coming across extremely slow data rates (~140Hz). We believe it to be something with the HAL library, so we are attempting to use this library as we have had success with it on the STM32F4 platform. 

The STM32L1XX has a simple USB peripheral instead of the USB_OTG_FS peripheral. My attempts at converting the code in this project to use the USB have been unsuccessful. We are using a verified report descriptor for the HID device (as in it works with the STM32 HAL library).

Here is the code we have changed.

**hal_usb.c**
```
#include "stm32l1xx_hal.h"

#include <xpd_usb.h>
#include <usb_device.h>

void HAL_USB_FS_MspInit(void* handle)
{
	PCD_HandleTypeDef hpcd_USB_FS;

	hpcd_USB_FS.Instance = USB;
	hpcd_USB_FS.Init.dev_endpoints = 8;
	hpcd_USB_FS.Init.speed = PCD_SPEED_FULL;
	hpcd_USB_FS.Init.phy_itface = PCD_PHY_EMBEDDED;
	hpcd_USB_FS.Init.low_power_enable = DISABLE;
	hpcd_USB_FS.Init.battery_charging_enable = DISABLE;
	HAL_PCD_Init(&hpcd_USB_FS);

	__HAL_RCC_USB_CLK_ENABLE();

	/* Peripheral interrupt init */
	HAL_NVIC_SetPriority(USB_HP_IRQn, 1, 0);
	HAL_NVIC_EnableIRQ(USB_HP_IRQn);
	HAL_NVIC_SetPriority(USB_LP_IRQn, 1, 0);
	HAL_NVIC_EnableIRQ(USB_LP_IRQn);
}

void HAL_USB_FS_MspDeInit(void* handle)
{
	__HAL_RCC_USB_CLK_DISABLE();

	/* Peripheral interrupt Deinit*/
	HAL_NVIC_DisableIRQ(USB_HP_IRQn);

	HAL_NVIC_DisableIRQ(USB_LP_IRQn);
}

void HAL_USBD_Setup(void)
{
    USB_INST2HANDLE(UsbDevice, USB);
    UsbDevice->Callbacks.DepInit = HAL_USB_FS_MspInit;
    UsbDevice->Callbacks.DepDeinit = HAL_USB_FS_MspDeInit;
}

void USB_IRQHandler(void);

void USB_IRQHandler(void)
{
	HAL_NVIC_GetPendingIRQ(USB_FS_WKUP_IRQn);

	USB_vIRQHandler(UsbDevice);
}
```

**gmk_controller_if.c**
```
#include <usbd_hid.h>
#include "stm32xxxx.h"
#include <stdio.h>
#include <string.h>

#define lowByte(w) ((uint8_t) ((w) & 0xff))
#define highByte(w) ((uint8_t) ((w) >> 8))

uint8_t ReportDescriptor[65] = {
	0x05, 0x01,                    // USAGE_PAGE (Generic Desktop)
	0x09, 0x05,                    // USAGE (Game Pad)
	0xa1, 0x01,                    // COLLECTION (Application)
	0xa1, 0x00,                    //   COLLECTION (Physical)
	0x85, 0x01,                    //     REPORT_ID (1)
	0x05, 0x09,                    //     USAGE_PAGE (Button)
	0x19, 0x01,                    //     USAGE_MINIMUM (Button 1)
	0x29, 0x10,                    //     USAGE_MAXIMUM (Button 16)
	0x15, 0x00,                    //     LOGICAL_MINIMUM (0)
	0x25, 0x01,                    //     LOGICAL_MAXIMUM (1)
	0x95, 0x10,                    //     REPORT_COUNT (16)
	0x75, 0x01,                    //     REPORT_SIZE (1)
	0x81, 0x02,                    //     INPUT (Data,Var,Abs)
	0x05, 0x01,                    //     USAGE_PAGE (Generic Desktop)
	0x09, 0x30,                    //     USAGE (X)
	0x09, 0x31,                    //     USAGE (Y)
	0x09, 0x32,                    //     USAGE (Z)
	0x09, 0x33,                    //     USAGE (Rx)
	0x16, 0x00, 0x80,              //     LOGICAL_MINIMUM (-32768)
	0x26, 0xff, 0x7f,              //     LOGICAL_MAXIMUM (32767)
	0x75, 0x10,                    //     REPORT_SIZE (16)
	0x95, 0x04,                    //     REPORT_COUNT (4)
	0x81, 0x02,                    //     INPUT (Data,Var,Abs)
	0x09, 0x34,                    //     USAGE (Ry)
	0x09, 0x35,                    //     USAGE (Rz)
	0x15, 0x00,                    //     LOGICAL_MINIMUM (0)
	0x26, 0xff, 0x00,              //     LOGICAL_MAXIMUM (255)
	0x75, 0x08,                    //     REPORT_SIZE (8)
	0x95, 0x02,                    //     REPORT_COUNT (2)
	0x81, 0x02,                    //     INPUT (Data,Var,Abs)
	0xc0,                          //   END_COLLECTION
	0xC0    /*     END_COLLECTION	             */
};

static const USBD_HID_ReportConfigType gmk_controller_hid_report =
{
		.Desc = ReportDescriptor,
		.DescLength = 65,
		.MaxId = 1,
		.Input =
		{
			.Interval_ms = 1,
			.MaxSize = 13,
		},
		.Feature =
		{
			.MaxSize = 13,
		},
		.Output =
		{
			.Interval_ms = 200,
			.MaxSize = 1,
		},
};

static const USBD_HID_AppType gmk_controller_app =
{
		.Name = "GMK HID Controller",
		.Report = &gmk_controller_hid_report,
};

USBD_HID_IfHandleType _gmk_controller_if = {
		.App = &gmk_controller_app,
}, *const gmk_controller_if = &_gmk_controller_if;
```

**usb_device.c**
```
#include <usb_device.h>
#include <usbd_cdc.h>
#include <usbd_hid.h>

/** @brief USB device configuration */
const USBD_DescriptionType hdev_cfg = {
    .Vendor = {
        .Name           = "GMK",
        .ID             = 0x0483,
    },
    .Product = {
        .Name           = "GMK Controller",
        .ID             = 0x5740,
        .Version.bcd    = 0x0100,
    },
    .Config = {
        .Name           = "GMK Controller Configuration",
        .MaxCurrent_mA  = 100,
        .RemoteWakeup   = 0,
        .SelfPowered    = 0,
    },
}, *const dev_cfg = &hdev_cfg;

USBD_HandleType hUsbDevice, *const UsbDevice = &hUsbDevice;

extern USBD_HID_IfHandleType *const gmk_controller_if;

void UsbDevice_Init(void)
{
    /* Configure HID controller */
    gmk_controller_if->Config.InEpNum = 0x83;

    /* Mount the interfaces to the device */
    USBD_HID_MountInterface(gmk_controller_if, UsbDevice);

    /* Initialize the device */
    USBD_Init(UsbDevice, dev_cfg);

    /* The device connection can be made */
    USBD_Connect(UsbDevice);
}
```

**usbd_config.h**
```
#define USBD_MAX_IF_COUNT           2

#define USBD_EP0_BUFFER_SIZE        256

#define USBD_HS_SUPPORT             0

#define USBD_SERIAL_BCD_SIZE        0

#define USBD_MS_OS_DESC_VERSION     0

#define USBD_CDC_NOTEP_USED         0

#define USBD_CDC_CONTROL_LINE_USED  0

#define USBD_CDC_BREAK_SUPPORT      0

#define USBD_DFU_ALTSETTINGS        0

#define USBD_DFU_ST_EXTENSION       0

#define USBD_HID_ALTSETTINGS        0

#define USBD_HID_OUT_SUPPORT        0

#define USBD_HID_REPORT_STRINGS     0
```

Upon plugging the device into the host computer, we get the error:
`A request for the USB device descriptor failed.`

If you have additional questions about the code or the device, please reach out and I will gladly provide you with some more information.

Thanks!

benedekkupper, 2023-01-19:
The `MspInit()` function is supposed to enable the necessary RCC clocks for all peripherals that are relevant, configure the GPIOs and enable the USB interrupt(s). You are missing the GPIO initialization, and you are also doing something you are not supposed to: calling into the HAL USB library (with `HAL_PCD_Init`). You also need to define the correct interrupt handler, if the interrupt number is called `USB_HP_IRQn`, then you will need a corresponding `USB_HP_IRQHandler()`, the `USB_IRQHandler()` in this case is not added to the interrupt vector table.

TannerHollis, 2023-01-24:
Hi, 
Thanks for your response, a couple of things.

1. We got it working.. Thank you very much. It was actually very simple.
2. I looked at STM's implementation of the peripheral initialization and they don't have a GPIO initialization, instead they call the `HAL_PCD_Init()`, which enables the peripheral. So I don't know a way around initializing the USB peripheral without calling that function.
3. All it took was adding the interrupt handlers to the interrupt vector table. `USB_HP_IRQHandler()` and `USB_LP_IRQHandler()`

Our implementation is available here: [https://github.com/TannerHollis/GMK_Joystick_2.0_XPD_USB](https://github.com/TannerHollis/GMK_Joystick_2.0_XPD_USB)

closed 2023-01-24 by TannerHollis