IntergatedCircuits/USBDevice4Cube #14: error: 'USB_CNTR_LPMODE' undeclared (first use in this function)

lazenyuk-dmitry, 2023-03-22:
I am not well versed in programming micro controllers and this is my first experience. I get a lot of compilation errors. I have corrected some of them. But I don't understand what to do with it.

[image]

benedekkupper, 2023-03-24:
Welcome to microcontroller programming, strap in, because this will be a bumpy ride! :D

It seems to me that you have also attempted to use TinyUSB, what stopped you there? That library sees a lot more use than this one, I'd think your chances are better with that.

Nonetheless, the error is most likely the result of renaming symbols in the MCU's CMSIS header (the one that goes like stm32f1xyz.h). This library was written several years ago, perhaps the vendor (ST) decided to change some names. My recommendation is, follow what the compiler recommends you. For simplicity, add compatibility defines like this:
#ifndef USB_CNTR_LPMODE
#define USB_CNTR_LPMODE USB_CNTR_LP_MODE
#endif 
To the stm32xxxx.h file. When it works, please send your changes via a pull request.

lazenyuk-dmitry, 2023-03-25:
Thanks for the answer) This is my joint project with a friend. So far, I don't even have an STM32 board for testing (waiting for delivery). Right now I'm trying to find a compact and simple library to implement HID USB. I tried CubeMX but there an empty HID USB project takes up 30 kilobytes of flash memory. I also tried tinyUSB and I was able to assemble it without errors. But the tinyUSB project takes less than ~20 kilobytes of flash memory. Why does the USB implementation on STM32 take so much memory ??? I would like to find the easiest solution.

benedekkupper, 2023-03-26:
All my knowledge is from years ago, but I'm pretty sure that ST's USB library isn't *that* heavy. A USB bootloader with this library fits in 8k flash, and ST's used to fit in 16k - with `-Os` size optimization. So I'm pretty sure that you can get much lower with the right optimization flags.

closed 2023-04-16 by lazenyuk-dmitry

benedekkupper, 2023-04-20:
What's the conclusion here, which library did you go with? I will keep the issue open until the F1 is ported.